BinghamtonRover/RoverSystem #100: Science Display and Data

layneson, 2019-02-04:
Design and prototype the GUI for the science display. Use a mockup tool of some kind (even pencil and paper) which indicates a rough idea of the overall layout. You have a 1920x1080 (16:9 aspect ratio) screen to work with, however we will want to display some status information (network status, rover location, etc.) , so leave room for a small bar across the bottom.

Make a list of the data that you would like to collect from the sensors and send to the base station. Indicate the data type and valid range of each measurement. This is extremely important information, as it will determine how science data is stored, transmitted, and processed.

Determine what processing needs to be done with the data in order for the display to give the science team as much helpful information as possible. For instance, maybe some sort of chart or graph will allow for easier interpretation of data.

JonathanWu1120, 2019-02-14:
Current values science team has
TBD values still need to be tested
Humidity - 0-100%
pH meter 3.5-9.0
Anomometer - TBD
Methane sensor - TBD
ATM pressure sensor - TBD
Co2 sensor - TBD

JonathanWu1120, 2019-02-21:
Humidity - 0-100%
pH meter 3.5-9.0
Anomometer - 0.5 m/s to 50 m/s
Methane sensor - 200 to 1000 ppm
ATM pressure sensor - 20 unsigned bit pascal value
Co2 sensor - PPM (0-2000+?)

closed 2019-09-18 by layneson