BinghamtonRover/RoverSystem #128: Emergency stop

AlexanderRBurke, 2019-09-26:
The rover has a big red emergency stop button on it which will shutdown the whole rover system manually. This button can be difficult to press if the rover is going full speed away from us. So your task is to implement a similar emergency stop through a software message which will shutdown the rover system remotely.

AlexanderRBurke, 2019-09-26:
[image] [Emergency stop](https://trello.com/c/djEFdQ38/12-emergency-stop)

jlopez81, 2019-10-03:
[Emergency Stop.pdf](https://github.com/BinghamtonRover/RoverSystem/files/3686667/Emergency.Stop.pdf)
[image]

[stop code.zip](https://github.com/BinghamtonRover/RoverSystem/files/3686672/stop.code.zip)

jlopez81, 2019-10-03:
[Emergency Stop.pdf](https://github.com/BinghamtonRover/RoverSystem/files/3686674/Emergency.Stop.pdf)
[stop code.zip](https://github.com/BinghamtonRover/RoverSystem/files/3686675/stop.code.zip)

layneson, 2019-11-04:
[image] [Emergency stop](https://trello.com/c/4ws1yxWw/40-emergency-stop)

closed 2021-03-23 by AlexanderRBurke